locka99/opcua #165: errors in configuration files are not reported

lovasoa, 2022-02-21:
When a configuration file cannot be deserialized, this is the only thing that is logged:

```
2022-02-21 17:55:37.608 - ERROR - opcua_core::config - Cannot deserialize configuration from ./server.conf
```

And the `ServerConfig::load` returns `Err(())`, so there is no way to know where the error is in the invalid configuration file.

locka99, 2022-02-21:
I'll check if there is any way to get meaningful errors out of serde but this is currently opaque - the code asks serde to  load and it either does or it doesn't.

lovasoa, 2022-02-22:
Yes, serde gives meaningful errors, that could be returned to the user, potentially wrapped in a custom error type. 
Also, serde_yaml can read a file directly, no need to read it to a string first: https://docs.rs/serde_yaml/latest/serde_yaml/fn.from_reader.html

locka99, 2022-02-23:
For now I've added code to log the error message that serde gives me into the error output

closed 2022-02-23 by locka99