locka99/opcua #166: asynchronous or blocking getters and setters

lovasoa, 2022-02-22:
Hello,

   I cannot find in the documentation whether value getters and setters are supposed to be non-blocking or not. Looking at the code, it looks like they are indeed supposed to be. And there seem to be no way to define an asynchronous variable getter or setter.

   What is the recommended way to implement a getter or setter that needs to interact with the network or the filesystem ?

locka99, 2022-02-23:
At present they're synchronous only. You could poll the current value and have your own thread in the background which refreshes that value.

lovasoa, 2022-03-02:
That would be very inefficient !

locka99, 2022-03-02:
Yes I agree. I think until async permeates through the api it would be the only way to do it.