locka99/opcua #170: client: Failure to decode server response leads to broken session

laumann, 2022-02-28:
This takes a little explaining, so I'll start with the steps to reproduce:

### Steps to reproduce

1. Run this docker string:
 ```$ docker run --rm -d --net host --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=50000 --ph=localhost --autoaccept --sph --sn=5 --sr=10 --st=uint --fn=5 --fr=1 --ft=uint --ctb --lsn --ref --gn=5 -ut```
2. Apply this patch: [0001-samples-Simple-client-to-read-nodes.patch.txt](https://github.com/locka99/opcua/files/8153696/0001-samples-Simple-client-to-read-nodes.patch.txt) on master (note I had to rename the patch generated by git to include the .txt, because github checks extensions... for some reason)
3. `cd samples/read-client`
4. `RUST_OPCUA_LOG=debug cargo run`

I get the following log:

<details>
<summary>output log</summary>

```
warning: field is never read: `max_chunk_count`
  --> core/src/comms/message_writer.rs:29:5
   |
29 |     max_chunk_count: usize,
   |     ^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: `#[warn(dead_code)]` on by default

warning: `opcua-core` (lib) generated 1 warning
warning: unused imports: `Arc`, `RwLock`
  --> samples/read-client/src/main.rs:10:17
   |
10 | use std::sync::{Arc, RwLock};
   |                 ^^^  ^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: `opcua-read-client` (bin "opcua-read-client") generated 1 warning
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
     Running `/home/tj/omnioiot/opcua/target/debug/opcua-read-client`
2022-02-28 13:58:36.480 - INFO - opcua_console_logging - Logging is enabled, use RUST_OPCUA_LOG environment variable to control filtering, logging level
2022-02-28 13:58:36.480 - WARN - opcua_client::config - Endpoint config contains no endpoints
2022-02-28 13:58:36.482 - INFO - opcua_client::session::session - Connect
2022-02-28 13:58:36.482 - INFO - opcua_client::session::session - Security policy = None
2022-02-28 13:58:36.482 - INFO - opcua_client::session::session - Security mode = None
2022-02-28 13:58:36.482 - DEBUG - opcua_client::comms::tcp_transport - Waiting for a connect (or failure to connect)
2022-02-28 13:58:36.483 - DEBUG - opcua_client::comms::tcp_transport - Client tokio tasks are starting for connection
2022-02-28 13:58:36.483 - DEBUG - opcua_core::runtime - registering component client-connection-thread-ThreadId(1)
2022-02-28 13:58:36.483 - DEBUG - opcua_client::comms::tcp_transport - Creating a connection task to connect to [::1]:50000 with url opc.tcp://localhost:50000
2022-02-28 13:58:36.483 - DEBUG - opcua_core::runtime - registering component connection-task, 1
2022-02-28 13:58:36.483 - DEBUG - opcua_client::comms::tcp_transport - Sending HELLO
2022-02-28 13:58:36.483 - DEBUG - opcua_core::runtime - deregistering component connection-task, 1
2022-02-28 13:58:36.483 - DEBUG - opcua_core::runtime - registering component finished-monitor-task, 1
2022-02-28 13:58:36.483 - DEBUG - opcua_core::runtime - registering component read-task, 1
2022-02-28 13:58:36.483 - DEBUG - opcua_core::runtime - registering component write-task, 1
2022-02-28 13:58:36.483 - DEBUG - opcua_client::comms::tcp_transport - Reader got ack AcknowledgeMessage { message_header: MessageHeader { message_type: Acknowledge, message_size: 28 }, protocol_version: 0, receive_buffer_size: 65535, send_buffer_size: 65535, max_message_size: 4194304, max_chunk_count: 0 }
2022-02-28 13:58:36.583 - DEBUG - opcua_client::comms::tcp_transport - Connected
2022-02-28 13:58:36.583 - DEBUG - opcua_client::session::session - session:1 open_secure_channel
2022-02-28 13:58:36.583 - INFO - opcua_client::session::session_state - Making secure channel request
2022-02-28 13:58:36.583 - INFO - opcua_client::session::session_state - security_mode = None
2022-02-28 13:58:36.583 - INFO - opcua_client::session::session_state - security_policy = None
2022-02-28 13:58:36.583 - DEBUG - opcua_core::comms::secure_channel - AsymmetricSecurityHeader = AsymmetricSecurityHeader { security_policy_uri: UAString { value: Some("http://opcfoundation.org/UA/SecurityPolicy#None") }, sender_certificate: ByteString { value: None }, receiver_certificate_thumbprint: ByteString { value: None } }
2022-02-28 13:58:36.583 - DEBUG - opcua_client::message_queue - Request 1 was processed by the server
2022-02-28 13:58:36.584 - DEBUG - opcua_client::message_queue - Response to Request 1 has been stored
2022-02-28 13:58:36.584 - DEBUG - opcua_client::session::session_state - Setting transport's security token
2022-02-28 13:58:36.584 - INFO - opcua_client::session::session - Connect was successful
2022-02-28 13:58:36.584 - DEBUG - opcua_client::session::session - session:1 get_endpoints
2022-02-28 13:58:36.584 - DEBUG - opcua_client::message_queue - Request 2 was processed by the server
2022-02-28 13:58:36.585 - DEBUG - opcua_client::message_queue - Response to Request 2 has been stored
2022-02-28 13:58:36.585 - DEBUG - opcua_client::session::session - session:1 get_endpoints, success
2022-02-28 13:58:36.586 - DEBUG - opcua_client::message_queue - Request 3 was processed by the server
2022-02-28 13:58:36.586 - DEBUG - opcua_client::message_queue - Response to Request 3 has been stored
2022-02-28 13:58:36.586 - ERROR - opcua_client::session::session - session:1 close_session failed ServiceFault(ServiceFault { response_header: ResponseHeader { timestamp: DateTime { date_time: 2022-02-28T13:58:36.586462900Z }, request_handle: 3, service_result: IS_ERROR | BadUnexpectedError | BadResourceUnavailable | BadCommunicationError | BadIdentityTokenInvalid | BadIdentityTokenRejected | BadNonceInvalid | BadSessionIdInvalid, service_diagnostics: DiagnosticInfo { symbolic_id: None, namespace_uri: None, locale: None, localized_text: None, additional_info: None, inner_status_code: None, inner_diagnostic_info: None }, string_table: Some([]), additional_header: ExtensionObject { node_id: NodeId { namespace: 0, identifier: Numeric(0) }, body: None } } })
2022-02-28 13:58:36.586 - ERROR - opcua_client - Received a service fault of BadSessionIdInvalid for the request
[... 60 lines cut ...]
2022-02-28 13:58:36.799 - INFO - opcua_client::comms::tcp_transport - ReadState has dropped
2022-02-28 13:58:36.799 - DEBUG - opcua_client::comms::tcp_transport - Writer write-task, 2 received a quit
2022-02-28 13:58:36.799 - DEBUG - opcua_client::comms::tcp_transport - Writer loop write-task, 2 is finished
2022-02-28 13:58:36.799 - DEBUG - opcua_core::runtime - deregistering component write-task, 2
2022-02-28 13:58:36.799 - INFO - opcua_client::comms::tcp_transport - WriteState has dropped
2022-02-28 13:58:36.800 - DEBUG - opcua_client::comms::tcp_transport - Connection state is finished so dropping out of connection task
```

</details>

Where I see the breakdown is the line "Unrecognized encoding mask"
```
2022-02-28 13:58:36.798 - ERROR - opcua_types::variant - Unrecognized encoding mask
2022-02-28 13:58:36.799 - DEBUG - opcua_core::comms::chunker - Cannot decode message ReadResponse_Encoding_DefaultBinary, err = IS_ERROR | BadUnexpectedError | BadInternalError | BadOutOfMemory | BadResourceUnavailable | BadCommunicationError | BadEncodingError | BadDecodingError
2022-02-28 13:58:36.799 - ERROR - opcua_client::comms::tcp_transport - Reader has put connection into a finished state with status BadServiceUnsupported
2022-02-28 13:58:36.799 - DEBUG - opcua_client::comms::tcp_transport - Sending a quit to the writer
```

For some reason, this shuts everything down and the read attempt eventually times out. But the session is broken in some way at this point, subsequent reads also time out.

In the code sample there are two nodes. `n1` produces this problem, but `n2` does not (a value is returned). Swapping the order of the nodes, first gives a value, then it crashes.

So far what I've figured is that `from_encoding_mask()` does not handle `Variant`. I don't know why this is, and I don't know if this is the real problem. What I would like though, is that a read that fails to decode is just returned to me as an error like everything else, so I, as the client, can decide on how to act. As it stands, the only way I can see to recover is to re-initialize the client, but that seems like overkill.

The signature of `from_encoding_mask()` was changed recently (in 633b60a1252719c84c879cbb60884652e4d9a7b1) and the change from a panic to returning an error looks like the right direction, but it doesn't fully solve this issue.

laumann, 2022-02-28:
I was also wondering if this is related to #145 (the type of the `n1` node is Variant)

laumann, 2022-02-28:
Sorry, I just realised the patch attached won't work on master. I'll rework it and post an update.

laumann, 2022-02-28:
[0001-samples-Simple-client-to-read-nodes.patch.txt](https://github.com/locka99/opcua/files/8154043/0001-samples-Simple-client-to-read-nodes.patch.txt)

This one should work.

schroeder-, 2022-02-28:
I think the problem is simple, currently there is no support for Variants containing other Variants. That is why you get "Unrecognized encoding mask", because in the EncodingMask case there is no match for Variant (24u8).

laumann, 2022-02-28:
Yes - I'm fine with the support lacking though. What I don't want is the client closing the connection when it fails to decode a server reply. I would like that to be handled more gracefully, by an Err(_) being returned to the caller.

I'm happy to author a patch and open a PR, but I think I would need some guidance on where the behavior needs changing.

schroeder-, 2022-02-28:
At this point the error is getting converted. For better diagnoses we should forward the error there.
https://github.com/locka99/opcua/blob/64844b9e6de7c2dbd8aacfa48a4c2d57aa3e96ac/core/src/comms/chunker.rs#L248-L262 

And here is the code where the connection is terminated, when decoding a message fails.
https://github.com/locka99/opcua/blob/64844b9e6de7c2dbd8aacfa48a4c2d57aa3e96ac/client/src/comms/tcp_transport.rs#L604-L619
Here there should be some recovery or better handling of the decoding error.

On the other hand I am working on a pr for implementing Variant::Variant.

locka99, 2022-02-28:
I've merged this so retest and see if it fixes the issue here. Weird for the server to nest variants in variants though

schroeder-, 2022-03-01:
Nested variants are used if you accept any datatype, or many datatypes, so you use BaseDataType as datatype for the variable.
My patch fixes only one problem. When there is a decoding error on the client side, the session will be terminated without the call knows about it and panic on the next request. And decoding errors can happen, for example if the server sends request that exceeds the decoding limits.

laumann, 2022-03-01:
I'll test it right away.

> When there is a decoding error on the client side, the session will be terminated without the call knows about it and panic on the next request.

This is what I'm trying to get at - is this the desired behaviour?

laumann, 2022-03-01:
> I've merged this so retest and see if it fixes the issue here. Weird for the server to nest variants in variants though

Tested and works, so I guess we can close this. I'll open another issue if I want to work on the "client failed to decode server response, session terminated" thing.

FWIW it's not a nested Variant types, it's an Array of Variant, I think. Here's the log output:

```
2022-03-01 08:52:14.037 - DEBUG - opcua_client::session::session - session:2 read() requested to read nodes [ReadValueId { node_id: NodeId { namespace: 5, identifier: String(UAString { value: Some("DataAccess_AnalogType_Array_Variant") }) }, attribute_id: 13, index_range: UAString { value: None }, data_encoding: QualifiedName { namespace_index: 0, name: UAString { value: None } } }]
2022-03-01 08:52:14.038 - DEBUG - opcua_client::message_queue - Request 4 was processed by the server
2022-03-01 08:52:14.242 - DEBUG - opcua_client::message_queue - Response to Request 4 has been stored
2022-03-01 08:52:14.242 - DEBUG - opcua_client::session::session - session:2 read(), success
Got value: [DataValue { value: Some(Array(Array { value_type: Variant, values: [Variant(Int32(10)), Variant(Int32(11)), Variant(Int32(12)), Variant(Int32(13)), Variant(Int32(14)), Variant(Int32(15)), Variant(Int32(16)), Variant(Int32(17)), Variant(Int32(18)), Variant(Int32(19))], dimensions: [] })), status: None, source_timestamp: None, source_picoseconds: None, server_timestamp: None, server_picoseconds: None }]
2022-03-01 08:52:14.242 - DEBUG - opcua_client::session::session - session:2 read() requested to read nodes [ReadValueId { node_id: NodeId { namespace: 5, identifier: String(UAString { value: Some("DataAccess_AnalogType_Array_UInt64") }) }, attribute_id: 13, index_range: UAString { value: None }, data_encoding: QualifiedName { namespace_index: 0, name: UAString { value: None } } }]
```
All the variants are then Int32, for whatever reason...

closed 2022-04-06 by laumann