locka99/opcua #179: Write bytes task is in error

lovasoa, 2022-03-24:
Since the latest update, when trying to connect to the opcua server from ignition (by inductive automation), the server logs the following error, and the connection fails. Nothing more is logged.

```
[2022-03-24T13:43:24.575Z ERROR opcua_server::comms::tcp_transport] Write bytes task is in error
```

locka99, 2022-03-25:
When you say latest update, which commit is that? I have updated the versions to 0.10 but there should be no changes to the code. However, one recent change requested was to loosen the tokio requirement from a specific version e.g. 1.18, to just 1.x.

lovasoa, 2022-03-25:
The bug started appearing when upgrading from 0.8.1 to 0.9.0: https://crates.io/crates/opcua-server/versions

It may be a consequence of https://github.com/locka99/opcua/issues/182 , the error appears in the logs after the unsuccessful connection from the java client, (the packet dump of which is in https://github.com/locka99/opcua/issues/182 ). It seems that this server isn't able to properly close the connection after the failed connection.

lovasoa, 2022-03-25:
You can reproduce it with a fresh milo client, for instance with https://hub.docker.com/r/inductiveautomation/ignition

PR #201 (2022-05-09): Fix memory leaks in connection management code
This revamps the connection handling on both the client and the server
to guarantee that we never have rogue tokio tasks running.

This simplifies the connection management code by using native rust and tokio
future cancelling to make sure all concurrent code stops when a connection dies,
whatever the reason.

This allows removing a lot of complex connection state management code.

In the new design, if either the read or the write loop needs to terminate the connection, it can do so by simply returning from the current function.

Fixes #199
Fixes [... cut]
commit 3ccdff87 (2022-05-09): Fix memory leaks in connection management code (#201)

* Fix memory leaks in connection management code

This revamps the connection handling on both the client and the server
to guarantee that we never have rogue tokio tasks running.

This simplifies the connection management code by using native rust and tokio
future cancelling to make sure all concurrent code stops when a connection dies,
whatever the reason.

This allows removing a lot of complex connection state management code.

Fixes #199
Fixes #179

* Fix clippy warnings

* Simplify client-side tcp session state management

Reduce the  [... cut]
closed 2022-05-09 by locka99