locka99/opcua #182: invalid nonce when opening secure channel

lovasoa, 2022-03-24:
When trying to connect to a rust opcua server from ignition, one gets the following error:

```
    UaException: status=Bad_NonceInvalid, message=nonce must be non-zero
``` 

Indeed, it looks like the server always sends a null string instead of a random nonce

https://reference.opcfoundation.org/v104/Core/docs/Part4/5.5.2/

lovasoa, 2022-03-24:
Here is a packet capture of the exchange between the "example server" from this repo and an out-of-the-box ignition client:

[pcap.zip](https://github.com/locka99/opcua/files/8344437/pcap.zip)

Here is a screenshot of what ignition shows:

[image]

[image]

lovasoa, 2022-03-24:
I tracked down this issue to https://github.com/locka99/opcua/commit/5c20ed8f1d3b708b5c0441528eea792a9607fa07

The server sends a nonce composed of 32 zeroes instead of the previous correct behavior of not including a nonce when the security policy is None.

locka99, 2022-03-25:
Yes I think this nonce stuff goes back and forth depending on the implementation being used against. The spec doesn't specify a nonce length and some client/server combinations want it and others do not

locka99, 2022-03-25:
I might revert it back to 0 but I'll have to review the change that put it to 32 bytes

lovasoa, 2022-03-25:
You can send a null  nonce to signal you don't have a nonce when the security policy is none, but sending an all-zero nonce is a big no no, in any situation.

locka99, 2022-03-25:
I meant a zero length nonce

closed 2022-04-05 by locka99