locka99/opcua #195: Client with 100% CPU usage

ctron, 2022-04-21:
Having a simple client example, subscribing to a single item, I have a CPU usage of 100%.

Maybe I am doing something wrong, or is that expected? I am mainly using the code of `simple-client`.

ctron, 2022-04-21:
I dug a bit into this, and couldn't find any difference to the `simple-client` example, which did work just fine.

I then ran an update on all dependencies (`cargo update`), and the problem was gone. The lock file had some old dependencies, which seem to have causes this.

:shrug: … sorry for the noise.

ctron, 2022-04-25:
So I did run into this again, even with newer dependencies. I dug a bit deeper, and noticed that it is repeatedly returning "none" in the following code:

https://github.com/locka99/opcua/blob/3b052d0f8a7c5fcb99f164ebb31c22d61c8a1d2d/lib/src/server/comms/tcp_transport.rs#L451-L453

Do my understanding, "none" would mean that the socket got closed. So I guess that should be treated as such, and shutdown reader and writer.

ctron, 2022-04-25:
So it looks like that this comes from a previous session, when connecting using `connect_to_endpoint`, where is does retrieve the remote endpoints. The connection gets closed, but the reader loop never gets cleaned up, endlessly polling the closed socket.

Adding a `break` to the situation, shutting down the writer in the process, seems to fix this issue.

@svanharmelen You might want to watch out for this too.

svanharmelen, 2022-04-25:
Thanks for the ping!

lovasoa, 2022-05-05:
This should be fixed in my pr : https://github.com/locka99/opcua/pull/201

@ctron : could you test it ?

You can specify in your dependencies

```
opcua = { git = "https://github.com/lovasoa/opcua", branch = "fix-zombie-tasks" }
```

ctron, 2022-05-06:
@lovasoa Sure, I will give it a try. I might however do this next Monday.

AiyionPrime, 2024-07-26:
@lovasoa @ctron As #201 has been merged and this should have been fixed by it, can this issue be closed?

lovasoa, 2024-07-26:
There was never a reproducible test for this issue, so I can't say for sure. But it should be. 

I think this can be closed.

ctron, 2024-07-26:
> @lovasoa @ctron As #201 has been merged and this should have been fixed by it, can this issue be closed?

Sorry, I didn't continue using it because of various reasons (partly because I no longer had the need for the use case). So I don't recall.

AiyionPrime, 2024-07-26:
@ctron I've put the client through its paces last week and it looked fine, so I'd suggest closing this until someone else were to find this was an issue, as it's supposed to be fixed?

locka99, 2024-07-31:
Closing per recommendation

closed 2024-07-31 by locka99