locka99/opcua #205: The secure channel lifetime is hardcoded to 1 minute

jonasar, 2022-05-13:
The requested secure channel lifetime is hardcoded as 60 seconds in function issue_or_renew_secure_channel() in session_state.rs.

This is a request to add this as a parameter to the client builder instead.

I am using version 0.8.1 of the opcua library, but I see that this is still hardcoded in the latest commit on master.

jonasar, 2022-05-13:
This short hardcoded secure channel lifetime makes https://github.com/locka99/opcua/issues/207 much worse since the connection is lost approximately every 3rd time (in my case, connecting to the OPC UA server in a Siemens Siematic S7-1500) that the secure channel token is renewed.

BogdanYarotsky, 2024-08-11:
@AiyionPrime hi, I'm looking for a first issue to contribute. Do you think it's a good idea if I try to tackle this one? Seems not that complicated on a first glance.

AiyionPrime, 2024-08-11:
Good afternoon @BogdanYarotsky, first things first, the crates author and maintainer is Mr. Lock.
He should be a way better addressee for your request, as I'm currently just a passer-by either.

That said, I haven't looked into this nor #207 yet.
Skimming through both issues I think resolving #207 would be the one more worthwhile, as this one appears to just make living with #207 more unpleasant.

I'm usually a fan of fixing the root of problems and not relieving it of its symptoms.

I'll have to look into the spec, whether configurable channel lifetimes would serve a (in my eyes/subjectively) more worthwhile purpose than relief of #207 to answer this more properly.

I'll try coming back to this within the next days unless @locka99 finds the time for an assessment.

BogdanYarotsky, 2024-08-11:
Good evening @AiyionPrime, thank you for the quick reply! I've checked the spec and it seems that making the secure channel lifetime configurable can be beneficial on it it's own. Here is a description of this lifetime from the spec (OPC UA Specification Part 4 - Services):

`
The requested lifetime, in milliseconds, for the new SecurityToken. It
specifies when the Client expects to renew the SecureChannel by calling
the OpenSecureChannel Service again. If a SecureChannel is not
renewed, then all Messages sent using the current SecurityTokens shall
be rejected by the receiver.

Several cryptanalytic attacks become easier as more material encrypted
with a specific key is available. By limiting the amount of data processed
using a particular key, those attacks are made more difficult. Therefore
the volume of data exchanged between Client and Server shall be
limited by establishing a new SecurityToken after the lifetime.

The setting of the requested lifetime depends on the expected number of
exchanged messages and their size in the lifetime. A higher volume of
data requires shorter lifetime.
`

The last part of the spec suggests to change the security channel lifetime duration based on the amount of data transferred.

I appreciate the link to #207, it's good to have a broader context of the usage, I will take a closer look of it in the following days. @locka99 please let know if you have any suggestions about this issue or possible solutions.

Have a nice week ahead!