locka99/opcua #207: Invalid signature reported on the received reply when renewing the secure channel

jonasar, 2022-05-13:
I am running version 0.8.1 of the opcua client.
Using security policy _Basic256Sha256_ and message security mode _SignAndEncrypt_.

When the secure channel token is renewed, approximately every 3rd time, the opcua client does not accept the response from the server, claiming that the data signature is invalid. 

Error "Signature invalid" is logged by function symmetric_verify_signature() in security_policy.rs (crypto/src/security_policy.rs).

This in turn leads to the connection being lost.

I am guessing that the opcua client switches to the new verification key immediately, although the OPC UA standard states that it shall accept messages secured by an expired security token for up to 25 % of the secure channel lifetime. 
(see https://reference.opcfoundation.org/v104/Core/docs/Part4/5.5.2/)

jonasar, 2022-05-13:
[opcua_client_signature_invalid.log](https://github.com/locka99/opcua/files/8686853/opcua_client_signature_invalid.log)

cailloumajor, 2023-03-30:
Hello, I encounter the same problem (see logs below) in my project https://github.com/cailloumajor/opcua-proxy.
The connection is made to an [IBH Link UA](https://www.ibhsoftec.com/IBH-Link-UA-Eng) OPC-UA server, which is very similar to Siemens SIMATIC NET v8.2 OPC-UA server.
I would be happy to help further.

[image]

BogdanYarotsky, 2024-08-16:
Hello @jonasar, hello @cailloumajor,
I would like to fix this issue. Is it still relevant for you?
Are you sure that the reason for this error is that security token is not accepted if it's expired but within the 25% duration of the secure channel lifetime?

cailloumajor, 2024-08-17:
> Hello @jonasar, hello @cailloumajor, I would like to fix this issue. Is it still relevant for you? Are you sure that the reason for this error is that security token is not accepted if it's expired but within the 25% duration of the secure channel lifetime?

Hello @BogdanYarotsky, unfortunately I'm not actively using this library anymore, so I fear I won't be able to help.

jonasar, 2024-08-17:
Hi @BogdanYarotsky <https://github.com/BogdanYarotsky>,

I am not actively using this library any more either.

Best regards,
Jonas