locka99/opcua #225: Relax protocol check

dsdjung, 2022-08-17:
Currently running version 0.10.0 against few commercially available server packages from
- Prosys
- UA Expert
- Integration Object

Connecting with UA Expert seems to work fine, but running against Prosys fails due to invalid Nonce error, and Integration Object due to BadCertificateUriInvalid error. Since both were working with earlier version of the library, I am guessing there has been some change in protocol handling/checks. 

I am wondering whether there are options to relax those checks when connection/sessions are created.

milgner, 2022-09-13:
Probably a duplicate of #150 - see https://github.com/locka99/opcua/issues/150#issuecomment-1245361332

pulzzedavid, 2022-09-14:
Looks like it. Looking forward to the PR.

dsdjung, 2022-09-14:
Looks like it. Looking forward to the PR.

closed 2022-09-14 by dsdjung