locka99/opcua #242: Replace chrono with time? (CVE-2020-26235)

0x2ec, 2022-12-01:
There's a known issue with chrono which apparently leads back to the dependency on time. One fix looks to be [updating chrono's dependency on time](https://github.com/chronotope/chrono/pull/632) from 0.1 to 0.3 but that doesn't appear [to have been done](https://github.com/chronotope/chrono/pull/639).

An alternative suggestion is to [replace chrono with time](https://github.com/meilisearch/milli/pull/450). Would it make sense to do this?

Another discussion: https://www.reddit.com/r/rust/comments/qqu1bw/what_should_we_do_about_cve202026235_localtime_r/

See:
 * https://nvd.nist.gov/vuln/detail/CVE-2020-26235
 * https://github.com/time-rs/time/security/advisories/GHSA-wcg3-cvx6-7396

antiblue, 2023-09-26:
Chrono dropped the dependency with [0.4.30](https://github.com/chronotope/chrono/releases/tag/v0.4.30), since `lib` currently references `chrono` in version 0.4 this should be okay now.

0x2ec, 2023-09-27:
Sounds right, thanks for the heads up! We'll close this.

closed 2023-09-27 by 0x2ec