locka99/opcua #256: Getting panic on otherwise successful reconnect

thosaa, 2023-03-16:
I am using the current master branch and I have a problem which is very alike #243.

I get a: 'main' panicked at 'MessageQueue::send_message should never be called before make_request_channel'
Which is due to hitting the expect at: https://github.com/locka99/opcua/blob/master/lib/src/client/message_queue.rs#L57

When I unplug our real PLC from the network, wait a bit and then reconnecting the PLC I get:
```
2023-03-16 14:46:41.744Z DEBUG opcua::client::message_queue - Request 21 was processed by the server
2023-03-16 14:46:53.743Z DEBUG opcua::client::session::session_state - async_publish, request sent with handle 22
2023-03-16 14:46:53.743Z DEBUG opcua::client::message_queue - Request 22 was processed by the server
2023-03-16 14:47:05.743Z DEBUG opcua::client::session::session_state - async_publish, request sent with handle 23
2023-03-16 14:47:05.743Z DEBUG opcua::client::message_queue - Request 23 was processed by the server
2023-03-16 14:47:14.593Z ERROR opcua::client::comms::tcp_transport - Read loop error Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
2023-03-16 14:47:14.593Z INFO  opcua::client::comms::tcp_transport - ReadState has dropped
2023-03-16 14:47:14.593Z INFO  opcua::client::comms::tcp_transport - WriteState has dropped
2023-03-16 14:47:14.593Z DEBUG opcua::client::comms::tcp_transport - Closing connection because the read loop terminated
2023-03-16 14:47:14.593Z DEBUG opcua::client::session::session_state - Session was closed with status = BadConnectionClosed
2023-03-16 14:47:14.603Z INFO  opcua::client::session_retry_policy - Retry retriggered by policy
2023-03-16 14:47:14.603Z INFO  opcua::client::session::session - Retrying to reconnect to server...
2023-03-16 14:47:14.603Z INFO  opcua::client::session::session - Connect
2023-03-16 14:47:14.604Z INFO  opcua::client::session::session - Security policy = Basic256Sha256
2023-03-16 14:47:14.604Z INFO  opcua::client::session::session - Security mode = SignAndEncrypt
2023-03-16 14:47:14.604Z DEBUG opcua::client::comms::tcp_transport - Creating a connection task to connect to 192.168.1.77:4840 with url opc.tcp://CX-61AEAE:4840/
2023-03-16 14:47:14.609Z DEBUG opcua::client::session::session - session:3 open_secure_channel
2023-03-16 14:47:14.609Z INFO  opcua::client::session::session_state - Making secure channel request
2023-03-16 14:47:14.609Z INFO  opcua::client::session::session_state - security_mode = SignAndEncrypt
2023-03-16 14:47:14.609Z INFO  opcua::client::session::session_state - security_policy = Basic256Sha256
thread 'main' panicked at 'MessageQueue::send_message should never be called before make_request_channel', /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/message_queue.rs:57:14
stack backtrace:
   0: rust_begin_unwind
             at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:575:5
   1: core::panicking::panic_fmt
             at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/panicking.rs:64:14
   2: core::panicking::panic_display
             at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/panicking.rs:147:5
   3: core::panicking::panic_str
             at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/panicking.rs:131:5
   4: core::option::expect_failed
             at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/option.rs:1924:5
   5: core::option::Option<T>::expect
             at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/option.rs:786:21
   6: opcua::client::message_queue::MessageQueue::send_message
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/message_queue.rs:56:22
   7: opcua::client::message_queue::MessageQueue::add_request
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/message_queue.rs:80:17
   8: opcua::client::session::session_state::SessionState::add_request
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session_state.rs:408:9
   9: opcua::client::session::session_state::SessionState::async_send_request
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session_state.rs:364:9
  10: opcua::client::session::session_state::SessionState::send_request
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session_state.rs:321:30
  11: opcua::client::session::session_state::SessionState::issue_or_renew_secure_channel
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session_state.rs:456:24
  12: <opcua::client::session::session::Session as opcua::client::session::services::SecureChannelService>::open_secure_channel
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session.rs:1305:9
  13: opcua::client::session::session::Session::connect_no_retry
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session.rs:531:13
  14: opcua::client::session::session::Session::reconnect_and_activate
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session.rs:304:13
  15: opcua::client::session::session::Session::poll::{{closure}}
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session.rs:729:24
  16: opcua::client::session::session::Session::session_task::{{closure}}::{{closure}}
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session.rs:632:39
  17: opcua::client::session::session::Session::session_task::{{closure}}::{{closure}}
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/macros/select.rs:505:49
  18: <tokio::future::poll_fn::PollFn<F> as core::future::future::Future>::poll
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/future/poll_fn.rs:38:9
  19: opcua::client::session::session::Session::session_task::{{closure}}
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session.rs:625:9
  20: opcua::client::session::session::Session::run_loop::{{closure}}
[... 12 lines cut ...]
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/coop.rs:99:5
  27: tokio::coop::budget
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/coop.rs:76:5
  28: tokio::runtime::basic_scheduler::Inner<P>::block_on::{{closure}}
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/runtime/basic_scheduler.rs:213:39
  29: tokio::runtime::basic_scheduler::enter::{{closure}}
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/runtime/basic_scheduler.rs:323:29
  30: tokio::macros::scoped_tls::ScopedKey<T>::set
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/macros/scoped_tls.rs:61:9
  31: tokio::runtime::basic_scheduler::enter
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/runtime/basic_scheduler.rs:323:5
  32: tokio::runtime::basic_scheduler::Inner<P>::block_on
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/runtime/basic_scheduler.rs:202:9
  33: tokio::runtime::basic_scheduler::InnerGuard<P>::block_on
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/runtime/basic_scheduler.rs:516:9
  34: tokio::runtime::basic_scheduler::BasicScheduler<P>::block_on
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/runtime/basic_scheduler.rs:162:24
  35: tokio::runtime::Runtime::block_on
             at /home/ts/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.8.4/src/runtime/mod.rs:450:46
  36: opcua::client::session::session::Session::run_loop
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session.rs:687:9
  37: opcua::client::session::session::Session::run
             at /home/ts/.cargo/git/checkouts/opcua-49af4c4898b3317f/ff2d148/lib/src/client/session/session.rs:584:9
 ...
```

thosaa, 2023-03-24:
Just a follow up.

It seems that there some session issues as the the MessageQueue.make_request_channel is called, but on the old session instead of the new session...

I have added a uuid to the MessageQueue, for debug purpose only, to be able to track the instances.
I also added debug logs starting with ### .

```
2023-03-24 08:46:44.260Z DEBUG opcua::client::session::session_state - async_publish is acknowledging subscription acknowledgements with sequence nrs [3]
2023-03-24 08:46:44.260Z DEBUG opcua::client::message_queue - ### send_message self = MessageQueue {
    inflight_requests: {
        13: None,
    },
    responses: {},
    sender: Some(
        UnboundedSender {
            chan: Tx {
                inner: Chan {
                    tx: Tx {
                        block_tail: 0x00007f53240008e0,
                        tail_position: 12,
                    },
                    semaphore: 0,
                    rx_waker: AtomicWaker,
                    tx_count: 1,
                    rx_fields: "...",
                },
            },
        },
    ),
    mq_id: 8bbeb5b0-adfe-4c8a-ab82-9d268076334a,
}
2023-03-24 08:46:44.260Z DEBUG opcua::client::session::session_state - async_publish, request sent with handle 13
2023-03-24 08:46:44.260Z DEBUG opcua::client::message_queue - Request 13 was processed by the server

///////// Here the connection was interrupted (Ethernet disconnected from PLC)  /////////

2023-03-24 08:46:56.469Z DEBUG opcua::client::message_queue - ### send_message self = MessageQueue {
    inflight_requests: {
        13: None,
        14: None,
    },
    responses: {},
    sender: Some(
        UnboundedSender {
            chan: Tx {
                inner: Chan {
                    tx: Tx {
                        block_tail: 0x00007f53240008e0,
                        tail_position: 13,
                    },
                    semaphore: 0,
                    rx_waker: AtomicWaker,
                    tx_count: 1,
                    rx_fields: "...",
                },
            },
        },
    ),
    mq_id: 8bbeb5b0-adfe-4c8a-ab82-9d268076334a,
}
2023-03-24 08:46:56.469Z DEBUG opcua::client::session::session_state - async_publish, request sent with handle 14
2023-03-24 08:46:56.470Z DEBUG opcua::client::message_queue - Request 14 was processed by the server

///////// Here the connection was retablished (Ethernet inserted)  /////////

2023-03-24 08:47:04.157Z ERROR opcua::client::comms::tcp_transport - Read loop error Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
2023-03-24 08:47:04.157Z INFO  opcua::client::comms::tcp_transport - ReadState has dropped
2023-03-24 08:47:04.157Z INFO  opcua::client::comms::tcp_transport - WriteState has dropped
2023-03-24 08:47:04.157Z DEBUG opcua::client::comms::tcp_transport - Closing connection because the read loop terminated
2023-03-24 08:47:04.158Z DEBUG opcua::client::session::session_state - Session was closed with status = BadConnectionClosed
2023-03-24 08:47:04.159Z INFO  opcua::client::session_retry_policy - Retry retriggered by policy
2023-03-24 08:47:04.159Z INFO  opcua::client::session::session - Retrying to reconnect to server...
2023-03-24 08:47:04.159Z DEBUG opcua::client::message_queue - ### MessageQueue new
2023-03-24 08:47:04.159Z DEBUG opcua::client::session::session_state - ### SessionState new has id: 3, and messag_queue id: e3ee6e23-bb8c-433d-a8c5-cdf72f695f49
2023-03-24 08:47:04.159Z INFO  opcua::client::session::session - Connect
2023-03-24 08:47:04.159Z INFO  opcua::client::session::session - Security policy = Basic256Sha256
2023-03-24 08:47:04.159Z INFO  opcua::client::session::session - Security mode = SignAndEncrypt
2023-03-24 08:47:04.160Z DEBUG opcua::client::comms::tcp_transport - Creating a connection task to connect to 192.168.1.77:4840 with url opc.tcp://CX-61AEAE:4840/
2023-03-24 08:47:04.161Z DEBUG opcua::client::comms::tcp_transport - ### WriteState new was hit, message_queue id: 8bbeb5b0-adfe-4c8a-ab82-9d268076334a
2023-03-24 08:47:04.161Z DEBUG opcua::client::message_queue - ### make_request_channel self = MessageQueue {
    inflight_requests: {},
    responses: {},
    sender: Some(
        UnboundedSender {
            chan: Tx {
                inner: Chan {
                    tx: Tx {
                        block_tail: 0x00007f5324019ad0,
                        tail_position: 0,
                    },
                    semaphore: 0,
                    rx_waker: AtomicWaker,
                    tx_count: 2,
                    rx_fields: "...",
                },
            },
        },
    ),
    mq_id: 8bbeb5b0-adfe-4c8a-ab82-9d268076334a,
}
2023-03-24 08:47:04.165Z DEBUG opcua::client::session::session - session:3 open_secure_channel
2023-03-24 08:47:04.165Z INFO  opcua::client::session::session_state - Making secure channel request
2023-03-24 08:47:04.165Z INFO  opcua::client::session::session_state - security_mode = SignAndEncrypt
2023-03-24 08:47:04.165Z INFO  opcua::client::session::session_state - security_policy = Basic256Sha256
2023-03-24 08:47:04.165Z DEBUG opcua::client::message_queue - ### send_message self = MessageQueue {
    inflight_requests: {
        1: Some(
            SyncSender { .. },
        ),
    },
    responses: {},
    sender: None,
    mq_id: e3ee6e23-bb8c-433d-a8c5-cdf72f695f49,
}
thread 'main' panicked at 'MessageQueue::send_message should never be called before make_request_channel', .../src/client/message_queue.rs:66:14

```

As the `### make_request_channel self =` shows the function is called and it holds a sender but the mq_id is the same  as before the connection was interrupted.

I will try to get wiser on the sessions and get back.

thosaa, 2023-03-24:
I fixed the panic with https://github.com/locka99/opcua/compare/master...thosaa:opcua:dont-panic-on-reconnect
But the code just stalls afterwards...

```
2023-03-24 10:48:46.679Z DEBUG opcua::client::session::session_state - ns=1;g=78589b75-5023-7741-b207-90801cb0ed85 Processing 1 async messages                                         
2023-03-24 10:48:46.679Z DEBUG opcua::client::session::session_state - ns=1;g=78589b75-5023-7741-b207-90801cb0ed85 handle_async_response                                               
2023-03-24 10:48:46.679Z DEBUG opcua::client::session::session_state - ns=1;g=78589b75-5023-7741-b207-90801cb0ed85 PublishResponse                                                     
2023-03-24 10:48:46.679Z DEBUG opcua::client::session::session_state - ns=1;g=78589b75-5023-7741-b207-90801cb0ed85 Received notifications, data changes = 1, events = 0                

///////// Storing the data from the PLC to the database /////////

2023-03-24 10:48:46.679Z DEBUG opcua::client::session::session_state - async_publish is acknowledging subscription acknowledgements with sequence nrs [7]                              
2023-03-24 10:48:46.679Z DEBUG opcua::client::session::session_state - async_publish, request sent with handle 17                                                                      
2023-03-24 10:48:46.680Z DEBUG opcua::client::message_queue - Request 17 was processed by the server  

////////// Disconnecting the PLC /////////

2023-03-24 10:48:59.118Z DEBUG opcua::client::session::session_state - async_publish, request sent with handle 18
2023-03-24 10:48:59.119Z DEBUG opcua::client::message_queue - Request 18 was processed by the server

////////// Reconnecting the PLC /////////

2023-03-24 10:49:06.579Z ERROR opcua::client::comms::tcp_transport - Read loop error Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
2023-03-24 10:49:06.579Z INFO  opcua::client::comms::tcp_transport - ReadState has dropped
2023-03-24 10:49:06.579Z INFO  opcua::client::comms::tcp_transport - WriteState has dropped
2023-03-24 10:49:06.579Z DEBUG opcua::client::comms::tcp_transport - Closing connection because the read loop terminated
2023-03-24 10:49:06.580Z DEBUG opcua::client::session::session_state - Session was closed with status = BadConnectionClosed
2023-03-24 10:49:06.589Z INFO  opcua::client::session_retry_policy - Retry retriggered by policy
2023-03-24 10:49:06.589Z INFO  opcua::client::session::session - Retrying to reconnect to server...
2023-03-24 10:49:06.589Z INFO  opcua::client::session::session - Connect
2023-03-24 10:49:06.590Z INFO  opcua::client::session::session - Security policy = Basic256Sha256
2023-03-24 10:49:06.590Z INFO  opcua::client::session::session - Security mode = SignAndEncrypt
2023-03-24 10:49:06.590Z DEBUG opcua::client::comms::tcp_transport - Creating a connection task to connect to 192.168.1.77:4840 with url opc.tcp://CX-61AEAE:4840/
2023-03-24 10:49:06.595Z DEBUG opcua::client::session::session - session:3 open_secure_channel
2023-03-24 10:49:06.595Z INFO  opcua::client::session::session_state - Making secure channel request
2023-03-24 10:49:06.595Z INFO  opcua::client::session::session_state - security_mode = SignAndEncrypt
2023-03-24 10:49:06.595Z INFO  opcua::client::session::session_state - security_policy = Basic256Sha256
2023-03-24 10:49:06.595Z DEBUG opcua::core::comms::secure_channel - AsymmetricSecurityHeader = AsymmetricSecurityHeader { security_policy_uri: UAString { value: Some("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256") }, sender_certificate: ByteString { value: Some([48, 130, 3, 199, ...]) }, receiver_certificate_thumbprint: ByteString { value: Some([223, 223, 227, 199, 30, 19, 120, 214, 53, 251, 134, 113, 11, 116, 43, 32, 74, 164, 125, 96]) } }
2023-03-24 10:49:06.596Z DEBUG opcua::client::message_queue - Request 1 was processed by the server
2023-03-24 10:49:07.757Z DEBUG opcua::client::message_queue - Response to Request 1 has been stored
2023-03-24 10:49:07.757Z DEBUG opcua::client::session::session_state - Setting transport's security token
2023-03-24 10:49:07.758Z DEBUG opcua::client::session::session - session:3 Endpoint policy = Some(UserTokenPolicy { policy_id: UAString { value: Some("Anonymous-[0]-SignAndEncrypt-Basic256Sha256") }, token_type: Anonymous, issued_token_type: UAString { value: None }, issuer_endpoint_url: UAString { value: None }, security_policy_uri: UAString { value: None } })
2023-03-24 10:49:07.760Z DEBUG opcua::client::message_queue - Request 2 was processed by the server
2023-03-24 10:49:07.762Z DEBUG opcua::client::message_queue - Response to Request 2 has been stored
2023-03-24 10:49:07.762Z ERROR opcua::client - Received a service fault of BadSessionIdInvalid for the request
2023-03-24 10:49:07.762Z INFO  opcua::client::session::session - Session activation failed on reconnect, error = BadSessionIdInvalid, so creating a new session
2023-03-24 10:49:07.762Z DEBUG opcua::client::session::session - session:3 create_session
2023-03-24 10:49:07.763Z DEBUG opcua::client::session::session - session:3 CreateSessionRequest = CreateSessionRequest { request_header: RequestHeader { authentication_token: NodeId { namespace: 0, identifier: Numeric(0) }, timestamp: DateTime { date_time: 2023-03-24T10:49:07.763119Z }, request_handle: 1, return_diagnostics: (empty), audit_entry_id: UAString { value: None }, timeout_hint: 10000, additional_header: ExtensionObject { node_id: NodeId { names [... cut ...]
2023-03-24 10:55:21.231Z INFO  opcua::client::comms::tcp_transport - ReadState has dropped
2023-03-24 10:55:21.231Z INFO  opcua::client::comms::tcp_transport - WriteState has dropped
2023-03-24 10:55:21.231Z DEBUG opcua::client::comms::tcp_transport - Closing connection because the read loop terminated
2023-03-24 10:55:21.231Z DEBUG opcua::client::session::session_state - Session was closed with status = Good
```

I was expecting the code to take up the previous subscription, what am I missing?

abdelhaksam, 2023-07-14:
The `run_loop` method didn't release the lock on the session runtime. I made a patch by running the session task on a fresh tokio runtime instead [here](https://github.com/abdelhaksam/opcua/tree/fix/subscription-deadlock-on-reconnect). It works now, but still requires more testing.