locka99/opcua #270: Discovering Object's references/components

abdawoud, 2023-06-05:
Hi,

I have the following node, which I got by traversing the root node:
```
let bd:[BrowseDescription;1] = [BrowseDescription {
    node_id: NodeId::new(2, 2),
    browse_direction: BrowseDirection::Forward,
    reference_type_id: ReferenceTypeId::Organizes.into(),
    include_subtypes: true,
    node_class_mask: 0,
    result_mask: BrowseDescriptionResultMask::all().bits() as u32
}];
```
It's an object, with some methods and attributes. Is there a way to extract it components and attributes?

Thank you.

schroeder-, 2023-06-05:
if you only need variables set:
`node_class_mask: NodeClassMask::VARIABLE `
if you only want objects set:
`node_class_mask: NodeClassMask::OBJECT`

or check the value `node_class` in `ReferenceDescription` in the response.

abdawoud, 2023-06-05:
Thanks for the reply and the hint. Unfortunately, I'm still unable to figure out the problem in my code. I would really appreciate it if you could help me find the issue:

```
let mut client = Client::new(client_config);

let endpoint: EndpointDescription = (
    "opc.tcp://127.0.0.1:4840/freeopcua/server/",
    "None",
    MessageSecurityMode::None,
    UserTokenPolicy::anonymous(),
).into();

match client.connect_to_endpoint(endpoint, IdentityToken::Anonymous) {
    Ok(session) => {
        let session = session.read();
        
        let bd:[BrowseDescription;1] = [BrowseDescription {
            node_id: NodeId::new(2, 2),
            browse_direction: BrowseDirection::Forward,
            reference_type_id: ReferenceTypeId::Organizes.into(),
            include_subtypes: true,
            node_class_mask: 0xff,
            result_mask: BrowseDescriptionResultMask::all().bits() as u32
        }];

        match session.browse(&bd) {
            Result::Ok(browse_result) => {
                println!("BrowseResult {:?}", browse_result);
            }
            Result::Err(status_code) => {
                println!("ERROR: {}", status_code);
            }
        }
    }
    Err(e) => {
        println!("{:?}", e);
    }
}
```
The results is:
`BrowseResult  Some([BrowseResult { status_code: HISTORICAL_RAW | Good, continuation_point: ByteString { value: None }, references: Some([]) }])`

 My objective is to read the red-circled references which are retrieved from UaExpert

[image]

schroeder-, 2023-06-05:
Use another  `reference_type_id` like `ReferenceTypeId::HasChild.into()`. 
See Figure 33 – Standard ReferenceType Hierarchy in https://reference.opcfoundation.org/Core/Part3/v105/docs/7

abdawoud, 2023-06-05:
Thank you very much! That solves my "issue". Much appreciated.