locka99/opcua #271: Simple Server Discovery Client Fails

guenhter, 2023-06-16:
When I run the simple-server by

```
cargo run
```

it starts up with some errors in the discovery:

```
2023-06-16 05:14:21.512 ERROR opcua::client::comms::tcp_transport      Could not connect to host 127.0.0.1:4840, Os { code: 111, kind: ConnectionRefused, message: "Connection refused" }
2023-06-16 05:14:21.512 ERROR opcua::client::session::session          session:1 Session has given up trying to connect to the server after 1 retries
2023-06-16 05:14:21.512 ERROR opcua::client::client                    Cannot connect to opc.tcp://localhost:4840/UADiscovery - check this error - BadNotConnected
2023-06-16 05:14:21.512 ERROR opcua::server::discovery                 Cannot find servers on discovery url opc.tcp://localhost:4840/UADiscovery, error = IS_ERROR | BadInternalError | BadEncodingLimitsExceeded | BadTimeout | BadTcpMessageTooLarge | BadTcpInternalError | BadSequenceNumberInvalid | BadNotConnected
```

After 5 minutes or so, the error repeats.

Any Idea why this is?

jcos1, 2023-11-09:
Remove the line 
`discovery_server_url: opc.tcp://localhost:4840/UADiscovery`
from `server.conf`

guenhter, 2023-11-14:
Thx,. This removed the errors.

closed 2023-11-14 by guenhter