locka99/opcua #284: Using simple server in Windows 10

caxelrud, 2023-08-23:
Hi, I got the following error when running simple-server example in Windows 10 Enterprise.
I also open the Firewall port 4840 for inbound and outbound but the error persisted.
I set environment variables for _DIR, _LIB_DIR, _INCLUDE_DIR and _CONF SSL_CERT_FILE to cacert.pem (downloaded from the internet).
Also, I included \bin to the PATH.

Please, let me know how solve this problem.

```
Running `C:\opcua-master\target\debug\opcua-simple-server.exe`
2023-08-22 21:33:29.588 ERROR opcua::client::comms::tcp_transport      Could not connect to host [::1]:4840, Os { code: 10061, kind: ConnectionRefused, message: "No connection could be made because the target machine actively refused it." }
2023-08-22 21:33:29.589 ERROR opcua::client::session::session          session:1 Session has given up trying to connect to the server after 1 retries
2023-08-22 21:33:29.591 ERROR opcua::client::client                    Cannot connect to opc.tcp://localhost:4840/UADiscovery - check this error - BadNotConnected
2023-08-22 21:33:29.592 ERROR opcua::server::discovery                 Cannot find servers on discovery url opc.tcp://localhost:4840/UADiscovery, error = IS_ERROR | BadInternalError | BadEncodingLimitsExceeded | BadTimeout | BadTcpMessageTooLarge | BadTcpInternalError | BadSequenceNumberInvalid | BadNotConnected
error: process didn't exit successfully: `C:\Users\CAxelru\OneDrive - Rockwell Automation, Inc\Research\RustOPCUA\opcua-master\target\debug\opcua-simple-server.exe` (exit code: 0xc000013a, STATUS_CONTROL_C_EXIT)
```

caxelrud, 2023-08-24:
Also, simple-client generates the following error:
```
2023-08-24 23:26:23.207 ERROR opcua::client::comms::tcp_transport      Could not connect to host [::1]:4855, Os { code: 10061, kind: ConnectionRefused, message: "No connection could be made because the target machine actively refused it." }
2023-08-24 23:26:25.248 ERROR opcua::client::comms::tcp_transport      Could not connect to host [::1]:4855, Os { code: 10061, kind: ConnectionRefused, message: "No connection could be made because the target machine actively refused it." }
2023-08-24 23:26:35.266 ERROR opcua::client::comms::tcp_transport      Could not connect to host [::1]:4855, Os { code: 10061, kind: ConnectionRefused, message: "No connection could be made because the target machine actively refused it." }
2023-08-24 23:26:35.267 ERROR opcua::client::session::session          session:1 Session has given up trying to connect to the server after 3 retries
2023-08-24 23:26:35.271 ERROR opcua::client::client                    Cannot get endpoints for server, error - BadNotConnected
```

digitizer0, 2023-09-27:
Have the same issue.

antiblue, 2023-09-29:
I have the same error messages, but I am still able to connect to the simple_server using UAExpert running on the same machine.
Is the same true for you @caxelrud & @digitizer0 ?

jcos1, 2023-11-09:
This might help:

Noticed that on my machine the simple-client tried to connect to localhost with IPv6 (::1) while the simple-server was listening on 127.0.0.1 (IPv4).
I bypassed it by modifying the `DEFAULT_URL` to `"opc.tcp://127.0.0.1:4855"` in simple-cliennt

I also had to remove this line from the server.conf because I have no discovery server running.
`discovery_server_url: opc.tcp://localhost:4840/UADiscovery`

LongZoz, 2023-11-21:
I also have the same problem with the default server code, although this does not affect the client to connect and read data, but how can this error be eliminated

jcos1, 2023-11-21:
> I also have the same problem with the default server code, although this does not affect the client to connect and read data, but how can this error be eliminated

Not really sure which problem you mean, but did you check this issue: #271

LongZoz, 2023-11-22:
Thank u,error magically disappeared