locka99/opcua #285: All TcpTransports sharing one SessionManager leads to unintended Session cancellation

thomasklunker, 2023-08-30:
Hi everyone,
I have run into a problem recently.

In commit 9741741 from May 25, 2022, the _Server_ class behavior was changed, so that all _TcpTransport_ instances it creates share one _SessionManager_ instance owned by the _Server_. Focus here on line 632:

https://github.com/locka99/opcua/blob/a5f452a67a0efbb1faaacda023612eea6d63be22/lib/src/server/server.rs#L626-L634

Now, once a _TcpTransport_ finishes, meaning when an OPC-UA client disconnects, it calls `TcpTransport::finish`:

https://github.com/locka99/opcua/blob/a5f452a67a0efbb1faaacda023612eea6d63be22/lib/src/server/comms/tcp_transport.rs#L311

This method in turn calls `SessionManager::clear` on its _SessionManager_ field, which, as I mentioned, is shared between all _TcpTransports_ (client connections). 

https://github.com/locka99/opcua/blob/a5f452a67a0efbb1faaacda023612eea6d63be22/lib/src/server/comms/tcp_transport.rs#L139-L140

`SessionManager::clear` terminates all _Sessions_, meaning all _Sessions_ from other _TcpTransports_ as well:

https://github.com/locka99/opcua/blob/a5f452a67a0efbb1faaacda023612eea6d63be22/lib/src/server/session.rs#L81-L90

This was not a problem before commit 9741741, when each _TcpTransport_ had its own separate _SessionManager_, so it would only close its own _Sessions_.
The way it is now, however, leads to the bug that I came upon and which is the reason I investigated this in the first place.

1. Start any (demo) OPC-UA server.
2. Connect 2 or more clients to that server.
3. Disconnect 1 client.
4. Try continue using any of the other clients. This fails with `BadSessionIdInvalid` since all sessions were closed.

I strongly suspect that this behavior is not intended. 
I managed to fix this issue by simply reversing the line change made in commit 9741741 that clones the _SessionManager_ _Arc_ into the _TcpTransport_, instead creating a new _SessionManager_ for each _TcpTransport_. Now, from the commit message of 9741741 I can see that having a shared _SessionManager_ is intentional, so this might not be the ideal solution to the problem. Maybe it would be preferrable to categorize the Sessions by _TcpTransport_ in the _SessionManager_ so that they can be closed separately.

Please tell me if I missed anything.

kevincolyar, 2023-08-30:
I wonder if this is the same issue I'm having with #281

thomasklunker, 2023-09-12:
> I wonder if this is the same issue I'm having with #281
I can definitely see them being related.

evanjs, 2024-03-08:
Pretty sure I'm encountering this lately
Feels like it could be a priority issue

Per the current [documented limitations](https://github.com/locka99/opcua/blob/master/docs/compatibility.md#current-limitations):
### Currently the following are not supported
* Multiple created sessions in a single transport.

If I understand the issue I/OP are experiencing, it almost seems like this "currently unsupported" scenario is (inadvertantly?) the way that the library currently functions, leading to the previously mentioned unintended session cancellation.