locka99/opcua #294: Mismatch Type Between Time and Chrono

NothinRandom, 2023-10-10:
https://github.com/locka99/opcua/blob/3610aceb72cad6d8dbd3ab39b26c56814b50dc9f/lib/src/server/subscriptions/monitored_item.rs#L263

Hello,

Sorry if this is a silly question, but I'm getting error, and I'm not sure why it's happening.  Currently running `cargo 1.73.0`:
```
   Compiling opcua v0.11.0
error[E0308]: mismatched types
   --> /home/test/.cargo/registry/src/index.crates.io-6f17d22bba15001f/opcua-0.11.0/src/server/subscriptions/monitored_item.rs:263:28
    |
263 |                 elapsed >= sampling_interval
    |                 -------    ^^^^^^^^^^^^^^^^^ expected `chrono::Duration`, found `time::Duration`
    |                 |
    |                 expected because this is `chrono::Duration`
    |
    = note: `time::Duration` and `chrono::Duration` have similar names, but are actually distinct types
note: `time::Duration` is defined in crate `time`

error[E0308]: mismatched types
   --> /home/home/.cargo/registry/src/index.crates.io-6f17d22bba15001f/opcua-0.11.0/src/server/subscriptions/subscription.rs:410:23
    |
410 |         if elapsed >= publishing_interval {
    |            -------    ^^^^^^^^^^^^^^^^^^^ expected `chrono::Duration`, found `time::Duration`
    |            |
    |            expected because this is `chrono::Duration`
    |
    = note: `time::Duration` and `chrono::Duration` have similar names, but are actually distinct types
note: `time::Duration` is defined in crate `time`

rror[E0308]: mismatched types
   --> /home/test/.cargo/registry/src/index.crates.io-6f17d22bba15001f/opcua-0.11.0/src/server/subscriptions/subscriptions.rs:321:63
    |
321 |             if now.signed_duration_since(request_timestamp) > publish_request_timeout {
    |                --------------------------------------------   ^^^^^^^^^^^^^^^^^^^^^^^ expected `chrono::Duration`, found `time::Duration`
    |                |
    |                expected because this is `chrono::Duration`
    |
    = note: `time::Duration` and `chrono::Duration` have similar names, but are actually distinct types
```

Simontaga, 2023-10-17:
I got around it by targeting the repo when including the dependency.

The latest version (v0.12.0) does not seem to be pushed to crates.io
```
[dependencies]
opcua = { git = "https://github.com/locka99/opcua.git", branch = "master", features = ["console-logging", "server"] }
```

NothinRandom, 2023-10-17:
I manually went into registry and changed time to chrono and fixed that just to be sure.  I guess strict check is getting more strict with each revision.  Anyway, pulling directly from repo since v12 doesn't exists in crates yet seems to fix the issue.  Hopefully, it will appear on there soon.  Now off to other mismatch types to fix... le sigh.

theodoreOnzGit, 2023-10-18:
Facing the same issue but the fix seems to work when pulling a bleeding edge version of opcua. Simontaga's solution works. Hope the new opc-ua version is put on crates io soon

NothinRandom, 2023-10-18:
@locka99 Thank you for your amazing contribution on this repo!

Is there any chance you could push v0.12.0 to crates to resolve some of these annoying issues?  It looks like the build tool that I'm using enforces a specific version, so pulling from git solves the mismatched types errors, but ultimately fails the build process.

NothinRandom, 2023-11-06:
@diconico07 thanks for the info!  Unfortunately, the internal build system that I'm dealing with is still throwing an error for not having being able to identify v0.12.0 in crates.

AiyionPrime, 2024-07-24:
The version has been published to crates.io 7 months ago:
https://crates.io/crates/opcua/0.12.0

@NothinRandom has your problem been resolved with the release of [0.12.0](https://github.com/locka99/opcua/releases/tag/0.12.0)?

NothinRandom, 2024-07-30:
@ AiyionPrime,  Yes.  Sorry, forgot to close this.

closed 2024-07-30 by NothinRandom