locka99/opcua #295: Broken docs.rs documentation. How to do a simple one time read without subscription?

kwinz, 2023-10-27:
Hi,

https://docs.rs/opcua/0.11.0/opcua/client/index.html shows how to start a subscription, but now how to one time poll/read/write a variable. Neither does the code in https://github.com/locka99/opcua/tree/master/samples

https://docs.rs/opcua/0.11.0/opcua/client/index.html  links to 

- https://docs.rs/opcua/0.11.0/opcua/client/client_builder/struct.ClientBuilder.html (The requested resource does not exist)
- https://docs.rs/opcua/0.11.0/opcua/client/session/struct.Session.html  (The requested resource does not exist)

Both of those links are broken. Probably the info how to read a variable was somewhere in that Session documentation?

I also tried https://docs.rs/opcua/0.10.0/opcua/client/session/struct.Session.html but that doesn't work either.

Ironically the only breadcrumb of an example how to do a one off read I found in an old closed issue https://github.com/locka99/opcua/issues/93 and obviously reading the source code helps a bit.

Can we fix the docs.rs documentation?

kwinz, 2023-10-30:
And how do I list **all** Nodes, DataNodes and Variables of the server?

I found `browse` but I am not sure what to use as `node_id` if I just want to browse everything.

```
let browse_description = [BrowseDescription {
    node_id: NodeId::new(2, 2), // ????
    browse_direction: BrowseDirection::Forward,
    reference_type_id: ReferenceTypeId::Organizes.into(), // ????
    include_subtypes: true,
    node_class_mask: NodeClassMask::all().bits(),
    result_mask: BrowseDescriptionResultMask::all().bits(),
}];

let browse_results = session.browse(&browse_description).unwrap().unwrap();

println!("BrowseResults {:?}", browse_results);
```

Some sample code would be really appreciated!

kwinz, 2023-10-30:
Besides the broken link in the README.md and not finding Session documentation.
I also created a question in discussions: https://github.com/locka99/opcua/discussions/296#discussion-5793457

mfwre, 2023-11-07:
You can find the Session documentation [here](https://docs.rs/opcua/latest/opcua/client/prelude/struct.Session.html).
It is indeed broken if you click the link from the `opcua::client` description. You can reach it following this route: `client > prelude > Session`.

In the Session documentation you'll find the [`AttributeService`](https://docs.rs/opcua/latest/opcua/client/prelude/struct.Session.html#impl-AttributeService-for-Session) trait which gives you the `fn read` method.

To do a simple read you can then:
```rust
let mut client = Client::new(config);
let session = client.connect_to_endpoint(..);
let values = session.read(&[ReadValueId], TimestampsToReturn:Both, 0f64);
```

Of course this is pseudo-code. Create an array of NodeId you want to get the value from, convert it into an array of ReadValueId (it implements `Into`) and pass it to the `read` method with some other parameters.

I hope this was useful.

kwinz, 2023-11-08:
Thank you @mfwre, much appreciated! I am not sure if you have time to also look at my question in the discussions https://github.com/locka99/opcua/discussions/296. Specifically how to list all available variable names/paths on a OPC UA server with this crate.

mfwre, 2023-11-09:
Hi, this is a more tricky answer.

The function you listed, `browse`, is the correct one.
If you have no clue on which nodes to start browsing on simply call the method [`NodeId::root_node_id()`](https://docs.rs/opcua/latest/opcua/types/node_id/struct.NodeId.html#method.root_folder_id) to get the server's root node.

As for what parameters to use in the function (#296: I'm referring to the `\\ ????` comments in your linked question) you'll find a detailed explanation in the OPC-UA specification; part 4 - section 5.8.2 and 5.8.3.

You still need to implement a "recursive" browsing yourself by repeating the `browse` call with different `NodeId`s.

I'll attach a screenshot of the page here for convenience.
[image]

AiyionPrime, 2024-07-25:
@kwinz I take it, the second half of this issue is resolved,
so all that's left here is to fix the dead links in the documentation?