locka99/opcua #316: error[E0277]: the trait bound `f64: From<bool>` is not satisfied

activatekillswitch, 2024-03-18:
Crate version 0.12.0 as well as on master e3dffb3

```
error[E0277]: the trait bound `f64: From<bool>` is not satisfied
   --> C:\...\.cargo\git\checkouts\opcua-49af4c4898b3317f\e3dffb3\lib\src\types\variant.rs:868:72
    |
868 |                     VariantTypeId::Double => Variant::Double(f64::from(v)),
    |                                                              --------- ^ the trait `From<bool>` is not implemented for `f64`
    |                                                              |
    |                                                              required by a bound introduced by this call
    |
    = help: the following other types implement trait `From<T>`:
              <f32 as From<i16>>
              <f32 as From<i8>>
              <f32 as From<u16>>
              <f32 as From<u8>>
              <f64 as From<f32>>
              <f64 as From<i16>>
              <f64 as From<i32>>
              <f64 as From<i8>>
            and 73 others
```

Both the f32 and f64 casts are failing compile

einarmo, 2024-03-20:
This was stabilized in rust 1.68, you're on a rather old rust version.

activatekillswitch, 2024-03-20:
Currently on
rustc 1.66.0 (69f9c33d7 2022-12-12)

Now
rustc 1.76.0 (07dca489a 2024-02-04)

Building perfectly!

closed 2024-03-20 by activatekillswitch