locka99/opcua #319: client/transport/state.rs: Private types in public interface errors

jpds, 2024-04-11:
These appear to have been introduced in #317:

```
   Compiling opcua v0.13.0 (https://github.com/locka99/opcua.git?branch=master#eab4caab)
error[E0446]: crate-private type `OutgoingMessage` in public interface
  --> /home/user/.cargo/git/checkouts/opcua-49af4c4898b3317f/eab4caa/lib/src/client/transport/state.rs:23:1
   |
23 | pub type RequestSend = tokio::sync::mpsc::Sender<OutgoingMessage>;
   | ^^^^^^^^^^^^^^^^^^^^ can't leak crate-private type
   |
  ::: /home/user/.cargo/git/checkouts/opcua-49af4c4898b3317f/eab4caa/lib/src/client/transport/core.rs:53:1
   |
53 | pub(crate) struct OutgoingMessage {
   | --------------------------------- `OutgoingMessage` declared as crate-private

error[E0446]: restricted type `client::transport::state::Request` in public interface
   --> /home/user/.cargo/git/checkouts/opcua-49af4c4898b3317f/eab4caa/lib/src/client/transport/state.rs:115:5
    |
42  |   pub(super) struct Request {
    |   ------------------------- `client::transport::state::Request` declared as restricted
...
115 | /     pub(crate) fn begin_issue_or_renew_secure_channel(
116 | |         &self,
117 | |         request_type: SecurityTokenRequestType,
118 | |         timeout: Duration,
119 | |         sender: RequestSend,
120 | |     ) -> Request {
    | |________________^ can't leak restricted type

For more information about this error, try `rustc --explain E0446`.
error: could not compile `opcua` (lib) due to 2 previous errors
```

CC: @einarmo

einarmo, 2024-04-11:
I can't actually reproduce this, what's your rustc version?

That said the error is clear enough: #320

jpds, 2024-04-11:
I got this error with cargo 1.73, but not with 1.75 or 1.76.

MSRV of the crate probably needs updating.

einarmo, 2024-04-11:
Interesting that it is only present in older rustc versions, a false positive maybe? We can just fix it. I'm on a newer rustc, but I'll try using a lower one when developing this project. I haven't been very careful about MSRV.

commit f448b14f (2024-04-12): Fix #319 (#320)

I can't actually reproduce this, but the error in the linked issue is
fairly clear, this should fix it.
closed 2024-04-12 by locka99