locka99/opcua #324: simple-client sample seems broken

cjjb01, 2024-05-01:
As the simple-server sample only publishes v1 and v2, the client is subscribed to v1,v2,v3,v4 and does not receive updates for v3,v4 and does nothing after the first update.
After changing the client to only subscribe to v1,v2 it behaves as expected and receives updates from the server.

AiyionPrime, 2024-07-02:
Hey @cjjb01; I just `git bisect`ed your problem, as I did not encounter it in `0.12.0`.
It appears the first commit introducing the bad behaviour is eab4caabab7b9bff4f23d5a35cfbc95c4327dc47, which was merged in #317.

@einarmo Can you reproduce the problem with your machine as well? Or is this unrelated to your (merged) PR?

einarmo, 2024-07-02:
I think I noticed this at one point and found it only happens with _this_ server SDK, so I wonder if it's a change in the client exposing a bug in the server, or it may be a bug in the client. I'll look into it when I find some time.

einarmo, 2024-07-16:
Alright. Whatever this is, it doesn't happen when running the client against other servers, or at all in my rewrite of the server part of the SDK (which finally runs the sample server now...).

AiyionPrime, 2024-07-16:
I'll see to verifying that tomorrow. Maybe I'll get that finding into the pipeline as well.

AiyionPrime, 2024-07-16:
Thanks for having looked into it that fast!