locka99/opcua #327: ARM musl cross-compile

ivansanchez-oss, 2024-05-23:
Thank you very much for this implementation, I am trying to compile for the armv7--unknown-linux-musleabihf target but I am having certain problems with openssl, I followed the steps in the documentation on cross compilation but it still does not work, if you suddenly have any ideas I would appreciate it quite

when I run ``cargo b --target=armv7-unknown-linux-gnueabihf`` everything works without problems but when I run ``cargo b --target=armv7-unknown-linux-musleabihf`` errors of undefined references appear

this is just a piece:

```
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: s3_cbc.c:(.text+0x7b4): undefined reference to `__memcpy_chk'
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: s3_cbc.c:(.text+0xa02): undefined reference to `__memset_chk'
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: /home/ivan/fun/arm-cod/target/armv7-unknown-linux-musleabihf/debug/deps/libopenssl_sys-1c964e63e2e24e60.rlib(eck_prn.o): in function `ECPKParameters_print':
          eck_prn.c:(.text+0x468): undefined reference to `__memset_chk'
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: eck_prn.c:(.text+0x4ee): undefined reference to `__memset_chk'
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: /home/ivan/fun/arm-cod/target/armv7-unknown-linux-musleabihf/debug/deps/libopenssl_sys-1c964e63e2e24e60.rlib(b_dump.o): in function `BIO_dump_indent_cb':
          b_dump.c:(.text+0x27a): undefined reference to `__memset_chk'
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: /home/ivan/fun/arm-cod/target/armv7-unknown-linux-musleabihf/debug/deps/libopenssl_sys-1c964e63e2e24e60.rlib(txt_db.o): in function `TXT_DB_read':
          txt_db.c:(.text+0x2b8): undefined reference to `__fprintf_chk'
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: /home/ivan/fun/arm-cod/target/armv7-unknown-linux-musleabihf/debug/deps/libopenssl_sys-1c964e63e2e24e60.rlib(dso_dlfcn.o): in function `dlfcn_name_converter':
          dso_dlfcn.c:(.text+0x1a2): undefined reference to `__sprintf_chk'
          collect2: error: ld returned 1 exit status

  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#rustc-link-lib)

```
``.cargo/config.toml``:

```
[target.armv7-unknown-linux-musleabihf]
linker="arm-linux-gnueabihf-gcc"               
```

```
[target.armv7-unknown-linux-gnueabihf]
linker="arm-linux-gnueabihf-gcc"       
```

variables: 
```
export OPENSSL_LIB_DIR=/tmp/openssl-1.0.1t/
export OPENSSL_INCLUDE_DIR=/tmp/openssl-1.0.1t/include
export OPENSSL_STATIC=1
```

ivansanchez-oss, 2024-05-23:
also tried the vendored-openssl feature but it only works for  ``--target=armv7-unknown-linux-gnueabihf``

locka99, 2024-05-27:
If you try to build openssl directly, i.e. checkout the openssl / openssl-sys repos directly and build them do you see a similar issue? 

I see there are some issues with openssl & musl which are reported, e.g. https://github.com/sfackler/rust-openssl/issues/603. Although that issue is closed it is referenced by a number of other projects dependent on openssl.

ivansanchez-oss, 2024-05-29:
I tried directly in the rust-openssl repository and I couldn't compile it either, it presented the same error but when I changed my distribution to Debian (before I was testing in Ubuntu) both this project and rust-openssl compiled without problem, I'm going to open an issue there and probably close this one, thanks for the attention!

closed 2024-06-05 by ivansanchez-oss