locka99/opcua #370: `drop` without `Drop`

AiyionPrime, 2024-07-26:
Hey @einarmo? What is the drop statement in line 73 supposed to do there?
Doesn't that only work on types implementing `Drop`?

The type of `activate_fut` is
`std::pin::Pin<&mut impl futures::Future<Output = std::result::Result<client::session::connect::SessionConnectMode, types::status_codes::StatusCode>>>`,

which is not `Drop`; does this do something I just don't get?

https://github.com/locka99/opcua/blob/8cc43e51fe2d14a965d271059aa031c89440fc30/lib/src/client/session/connect.rs#L44-L73

einarmo, 2024-07-26:
I think I did this because it didn't compile at some point, but it seems fine now, I can't remember why I did it now. You sometimes get this kind of error, where the default drop-order of values in a method causes a compiler error. The compiler is almost always smart enough to realize that it can drop values early, but you occasionally get an error which requires you to manually drop a value before the end of the function. That might have happened here.

As for your question, `std::mem::drop` works fine on types that don't explicitly implement `Drop`. Every type has a destructor, which recursively calls drop on all its members, `drop` ends up invoking that.

`Drop` isn't the destructor, it's more like a callback that runs before the destructor runs. Confusingly we just call destroying a value for "dropping" in rust.

einarmo, 2024-07-26:
This specific instance of `drop` can be removed, unless it causes issues.

PR #371 (2024-07-27): feat: Drop expendable drop
> as it only extends lifetimes at this point and is likely a refactoring artifact.

If this does not break everything, this closes #370.
Thanks @einarmo for the fast explanation in the issue!
closed 2024-07-27 by locka99