locka99/opcua #414: Community building

HerrMuellerluedenscheid, 2024-11-19:
Hi @locka99 ,

You've build a truely awesome library which is getting pretty popular given the number of stars and forks! At the same time I see that there is quite a number of open pull requests and I was wondering if you could need some help by the community to review, approve and possibly even merge pull requests. Are there already more reviewers then yourself on board for that? Would you be willing to share the workload? I think that the iteration of developments would benefit from that.

Also, I'm always a bit worried about software legacy. I've seen too many packages with single maintainers that for unknown reasons suddenly were not to be seen anymore leaving unmaintained packages in package managers (which is really a pitty but can also become quite dangerous if people continue relying on that). Yes, you can always branch, upload a crate `opcua-new` or `opcua-2` to crates.io but that's never an elegant solution.

What are your thoughts about that? 

ps.: I know, nobody wants to talk about the **real** end, but you might also consider setting up a [successor](https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-access-to-your-personal-repositories/maintaining-ownership-continuity-of-your-personal-accounts-repositories#inviting-a-successor) for your github account. I don't want to be gaslighting. It's just nice to have things in order :)

HerrMuellerluedenscheid, 2024-11-19:
I see that e.g. @AiyionPrime contributed recently and has open merge requests. Maybe you would be a good candidate? Also @svanharmelen, you were contributing quite a bit. Just a shoutout trying to get people involved ;)

einarmo, 2024-11-19:
I've got a fork that I've been working on with a pretty enormous amount of changes, at this point I have touched pretty much every part of the code in one way or another. I'll eventually finish doing everything I want to with this library, at which point I would like to somehow share it. _Not_ sharing it isn't really an option for me, considering the amount of work that has gone into it at this point...

What @locka99 has done here is great, and without the foundation he and others built I would've never gotten as far as I have, but at this point I think it's clear that he doesn't want to be involved as deeply in this project as it will require if he is to be the sole maintainer for the _only_ active OPC-UA project in Rust.

I only see two real ways forward:

 - Fork and publish with a different name. It would not be the first time this has happened in an open source project, and it won't be the last, but obviously it would be best to avoid...
 - Create a github org and move this repo there, then add at least two maintainers from the community. I'm willing, and @svanharmelen is willing, perhaps others that have contributed would be willing as well.

I really don't want to make this an ultimatum, like "make me maintainer or I fork", but I also feel like it's productive to lay the cards on the table. This project _is_ going to keep getting contributions, I'm pretty sure, and it does have issues that needs resolving (I've found and fixed quite a few in my time working on my fork).

If you don't want to give away the project, or share influence over it, (a sentiment I can understand...) then perhaps a coordinated fork would be for the best.

HerrMuellerluedenscheid, 2024-11-19:
Thanks for both your inputs! Awesome @einarmo that you brought the project forward a lot as it seems!

Definately the more pragmatic and sustainable way and clearly my personal favorite:

> Create a github org and move this repo there, then add at least two maintainers from the community. I'm willing, and @svanharmelen is willing, perhaps others that have contributed would be willing as well.

@locka99 what do you think? Looking really much forward hearing your thoughts!

AiyionPrime, 2024-11-21:
While I second @svanharmelen's suggestion to include @einarmo in a list of useful maintainers,
I do not think putting this repo in the hands of a (newfound-)org is in his interests, iirc he had a debate with @oroulet about the matter a while ago, which did not help either of them.

I've got a bunch of improvements at our fork as well, of which I promised @locka99 to only open five at a time in order to keep the PR overhead at bay.

Anyway, if @locka99 found a way to get this repos development moving again, I'd be quite happy about it as well. Just keep in none of us has a clue how his priorities are at the moment, let alone why; I'd refrain from ultimatums or fork announcements, as they're imo not really the best grounds to have an open discussion about this projects furture and @locka99's visions for it.

oroulet, 2024-11-21:
As I wrote earlier I used to maintain some OPC ua c++ stack and now maintain the opcua-asyncio stack using the freeopcua organisation. 
As I proposed earlier I can fork the project under that organization and let any of you become maintainer so we are not depending on one person.
As far as I can see the original maintainer here is not contributing much anymore while some of you do so I am ready to do that just now. Just propose a good name and I ll do it.
Also that organization is not affiliated to any company and I can give admin access to anyone showing good understanding of us and code. 
Of course you can make a new organization but I think putting together different stacks is not a bad idea.

I can start by proposing a bad name opcua2 or rust-opcua. Anyone has a better idea?

oroulet, 2024-11-21:
Also if anyone does not trust me they can just search my record in maintaining that organization. I have no other motivation than trying to keep projects alive and well functioning

AiyionPrime, 2024-11-21:
Now that was a fast reaction :)

All I wanted to express above was my urgent wish that we all keep our heads down for a moment and do **not** rush to possible solutions, before we have not even heard @locka99's current stand on the matter.

@oroulet I understand you only want to help here, but it's not that we're not incapable of forking this, but do intend to avoid it if possible and work on a solution **with** the current maintainer and not against him.

Thanks for your ongoing interest though, maybe we can just give @locka99 a few days to find time for an answer, before we escalate this further :)

HerrMuellerluedenscheid, 2024-11-22:
Wow! That sparked a discussion. Seems I hit a nerve :) Cool!

I saw [this discussion](https://github.com/locka99/opcua/discussions/306) which indicates that you would rather not move that project, which I can relate to. (though integrating that into the [FreeOpcUa](https://github.com/FreeOpcUa) group would probably boost visibility, community adoption and reputation of the project and as a consequence - winwin 🏆 ).

But foremost as long as we can spread some workload I would be super happy @locka99

oroulet, 2024-11-22:
Just so I have written it. After having worked with different open source projects for 20 years I think organization is the only way to ensure perennity of a project. Anything else is too much depending on one person which almost always end up disappearing or half disappearing like happening here.

oroulet, 2024-11-26:
has any of you created something to facilitate method creation btw? I just looked at the API now and it is quite heavy... Might be worth doing something

oroulet, 2024-11-30:
still not ready for doing something @AiyionPrime @einarmo or others?
I have a bunch of things I would like to improve like that one: https://github.com/locka99/opcua/pull/417  but I need a living project  to do that...

einarmo, 2024-11-30:
I have this: https://github.com/einarmo/opcua/tree/rewrite-master, which is really far ahead of master at this point. If nothing happens here, once I have done everything I want to, I'll probably make the effort to detach the fork and make it official, then we can see about reconciling things later.

oroulet, 2024-11-30:
OK I see you changes many things and also stuff I want. do you accept MR? Can I send my MR there then?
Also why waiting before publishing under an org? nothing is happening here anyway and this is frustrating

einarmo, 2024-11-30:
Feel free. I have changed pretty much everything, though, and I have my own vision for some of the core stuff. The reason I haven't gone ahead and released is that I would _prefer_ not to split the community, and I don't feel like what I have is completely ready for public consumption yet. There's a certain level of polish and robustness that I don't feel confident I have yet.

Your cancellation token PR is obsolete though, I think. There are ways to set a cancellation token when creating the server in my branch, and a bunch of stuff around cancellation, like logic to correctly shut down a server gracefully, and notifying clients with the `SecondsToShutdown` server variable.

oroulet, 2024-11-30:
I do not see publishing a fork as splitting the community. First there is no community here  anyway and then there is no problem to merge back things if we want.
The only issue is when someone like you changed everything but that problem is here anyway with or without fork. 

Yes when I saw your commits I understood my changes were obsolete. Fine, I can have a look at your fork.

AiyionPrime, 2024-11-30:
I'd personally favor waiting for @locka99's response.
But that's easy for me to say, I know everything I need for professional use is implemented already or just a few commits away.

For me there won't be going to much upstream changes until January, so waiting for an answer is not that much trouble anyway.
I'm not that much afraid of splitting the community, but was previously under the impression @locka99 did a decent job and would be missing his work in a fork.

But that's liek I said easy for me to say :)

einarmo, 2024-12-01:
As I said, if nothing happens I'll get around to doing that eventually. There's a little work involved in properly detaching the fork, and I want to go over documentation for all my changes, finish up some of the stuff I've started, and maybe expand on the tests a bit.

If I'm going to publish my own fork, then I want to to be high quality.

oroulet, 2024-12-01:
Don't be so careful. If you publish you will get help from people to make things good quality. I got help on projects in crazy bad shape

einarmo, 2024-12-19:
For the record, my fork is now published under https://github.com/FreeOpcUa/rust-opcua. There is still more work to be done to actually publish a new version to crates.io under a different name.

svanharmelen, 2024-12-19:
Thanks, I'll head over and take a look... Glad to see that its still published under MPL as I was a bit worried about that since most other repo's in that org use LGPL 👍🏻

svanharmelen, 2024-12-19:
Noticed you already pulled in a few of my PR's (al least the most important ones), will check if the others still make sense and open new PR's for them if needed so I can try to migrate our code base and do a few tests to see if the changes between the current client code here and in the new fork are compatible with how we use the client (we only use the client bits for now). I'll continue our communication at the new repo...

AiyionPrime, 2024-12-19:
I'll provide my work at the new repo as well.
@locka99 if you find time to address this matter though, just let us know and I'm sure we'll find a solution that suits all of our needs and interests.

oroulet, 2024-12-20:
great to see activities. @einarmo is the boss on that repository, but the idea is to add several people who want to help as part of the group. So if you want to contribute, we can invite you so you can help reviewing and merging things.